d4rk50ul1/klipper-on-android #7: After executing this script, Moonraker stopped starting

muraveix, 2022-10-29:
scripts/data-path-fix.sh

[image]

ZerGo0, 2022-10-29:
Just redo the following steps:
```
sudo wget -O /etc/default/moonraker https://raw.githubusercontent.com/d4rk50ul1/klipper-on-android/main/scripts/etc_default_moonraker
sudo wget -O /etc/init.d/moonraker https://raw.githubusercontent.com/d4rk50ul1/klipper-on-android/main/scripts/etc_init.d_moonraker

sudo chmod +x /etc/init.d/moonraker

sudo update-rc.d moonraker defaults
```
Do the same for Klipper if you have the same issue there

muraveix, 2022-10-30:
Thank you, the problem is solved.

closed 2022-10-30 by muraveix