d4rk50ul1/klipper-on-android #9: Stuck

SkyShadex, 2022-11-01:
Day 3 of trouble shooting.

So, flashed the Ender 3 v2 4.2.2. Screen is black so that's all I can confirm on that end.

- I'm able to find /dev/pts/3 via octo4a & termux.
- Pulled up the config in winscp and edited them
       - winscp because it just wouldn't let me do it via Fluidd
       -  `serial: /dev/serial/by-id/<my id>` replaced with  `serial: /dev/pts/3`
- In Fluidd, continues to say klipper service not connected.
       - When restarted it says klipper service not installed
       - Restarted it in ssh and it's on

The only things I've done since are let updates run through fluidd since it was reading klipper and moonraker as invalid
I can edit the configs directly in fluidd now. But klipper is still nowhere to be found.

currently testing this out https://klipper.discourse.group/t/klipper-service-not-found/864/6

SkyShadex, 2022-11-01:
that link was not the answer.

SkyShadex, 2022-11-01:
It would seem that moonraker is looking in `/home/android/klipper_config/` for files. But when I LS, that file does not exist. There's klipper, klipper_env, klippy. But no klipper_config. 

Furthermore, the file its looking for exists in /printer_data... But it's not looking where it should be. So I assume that's why it thinks klippy is missing/not installed?

SkyShadex, 2022-11-01:
Thinking my issue is that I did not do any of the normal klipper install steps.

I did everything through kiauh, including building the config. I imagine that script would run the `make` steps.

I believe the only part I'm confused about is how `/dev/pts/3` should replace `/dev/serial/by-id/`.

Should I still be running some version of `/by-id/*`

d4rk50ul1, 2022-11-01:
Have you gone through all the latest script updates?, i´m sorry i´m not right now at home and cannot ssh to help you troubleshoot but @ZerGo0  mentioned some fixes on the latest kiauh deployment. seems like some paths have changed.
now regarding the printer.cfg part on /dev/pts/3, it depends on if your printer is recognized using lsub or not. thing here is that some cellphones have the serial driver activated on the kernel, some just not no matter if you flash a different rom. Octo4a does a sort of driver implementation that allows you to find printer using the /dev/pts/#.

SkyShadex, 2022-11-01:
Hey! Thanks, also not at home. Just brainstorming. Will probably do a clean wipe and try it all again

Maybe? I checked the updates in kiauh (kiauh also had an update yesterday I believe so I ran that), said it was up to date. But in fluidd it listed moonraker and klipper as invalid. It would update them though. I'm unsure what steps I took but eventually got fluidd to push the updates. Other than that I'm unsure what else I should be doing.

I did try to run your suggested data-path-fix.sh but it fails when it gets that part. I don't remember the error (will update later). iirc, it's trying to patch a location that doesn't exist.

-------------------------------

Yeah I used Octo4a and termux to find to /dev/pts/3. When I 'ls /dev/', I never saw any /serial, /ttyA... or /ttyUSB0. So i figured I was missing that driver and went to Octo4a. Saw Octo4a was connected to the printer, went to termux 'sudo ls -al /data/data/com.octo4a/files/serialpipe' and /dev/pts/3 was what I arrived at.

- So if I'm correct, printer.cfg should read '[mcu] serial: /dev/pts/3' and then give that to klipper 'cd klipper; sudo chmod 777 /dev/pts/3', and I'm done?
- Or I'm also supposed to follow the normal klipper install steps of finding a serial id with dev/pts/3? 
- Other thing I just thought of, maybe octo4a is holding onto the port exclusively in the background? I'll have to check to make sure that's not the case.

-----------------------

As far as what's being recognized. It's an ender 3 v2 4.2.2 mcu. Below are the settings I followed, I went with option 1. Finished with the SD card flash using winscp to grab the bin. The phone is a moto x pure running lineage os 17.1. Are you saying, despite octoprint being able to find it, some phones just won't because of that missing driver?

>To use this config, during "make menuconfig" select the
>STM32F103 with a "28KiB bootloader" and serial (on USART1 PA10/PA9)
>communication.
>
>If you prefer a direct serial connection, in "make menuconfig"
>select "Enable extra low-level configuration options" and select
>serial (on USART3 PB11/PB10), which is broken out on the 10 pin IDC
>cable used for the LCD module as follows:
>3: Tx, 4: Rx, 9: GND, 10: VCC
>
>Flash this firmware by copying "out/klipper.bin" to a SD card and
>turning on the printer with the card inserted. The firmware
>filename must end in ".bin" and must not match the last filename
>that was flashed.

ZerGo0, 2022-11-01:
Hey, Moonraker changed the paths a few weeks ago, I'm pretty sure that they added a bunch of checks so that the fix.sh script works even if some paths don't exist after my feedback. I did sym linked everything myself, but they told me that.

You also need to add this to your `moonraker.cfg`:
```
[machine]
validate_service: False
validate_config: False
provider: none
```
They are using a bunch of systemd stuff now which doesn't work in Linux Deploy. The changes above pretty much use the old system and after that you should be able to connect to Klipper again and the update status should work as well.

SkyShadex, 2022-11-01:
Quick readup on symlink, that's pretty cool. Now to find my missing paths.

ZerGo0, 2022-11-01:
> I would love to learn how to sym link it myself, if you could point me in the right direction.

A sym link is pretty much just like a shortcut. This is how you do it: `ln -s /target/path/to/point/to/ /path/that/points/to/it/`

SkyShadex, 2022-11-01:
All about saving time. Here are my configs. I haven't started the clean wipe yet.
I usually do so by removing the environment in linux deploy.

[fluidd2.txt](https://github.com/d4rk50ul1/klipper-on-android/files/9914181/fluidd2.txt)
[printer2.txt](https://github.com/d4rk50ul1/klipper-on-android/files/9914182/printer2.txt)
[moonraker2.txt](https://github.com/d4rk50ul1/klipper-on-android/files/9914183/moonraker2.txt)

SkyShadex, 2022-11-01:
this is the error I get with the data-path-fix.sh

"Error: unable to find config: /home/android/klipper_config/moonraker.conf"

SkyShadex, 2022-11-01:
[image]
tried it in both directions and didn't get anywhere.

ZerGo0, 2022-11-02:
`klippy_uds_address` is definitely not correct in your Moonraker config, we use the default one and you don't need to define it.

Check the Klipper status with: `sudo /etc/init.d/klipper status` make sure that it is actually running.

ZerGo0, 2022-11-02:
> [image] tried it in both directions and didn't get anywhere.

You don't need to link the individual files, only the paths. I'm pretty sure that you have all your configs in the right directory already (/home/android/printer_data/config) which means that you don't have to run the `data-path-fix.sh` nor sym link anything.

SkyShadex, 2022-11-02:
> `klippy_uds_address` is definitely not correct in your Moonraker config, we use the default one and you don't need to define it.

I didn't define it. Even on today's clean install it started like that. Working to clean it up.

Progress: 
[image]

Moonraker found klippy. For some reason the `klippy_uds_address` defaulted to `klippy_uds_address = /home/android/printer_data/comms/klippy.sock` in my install. It should be `klippy_uds_address: /tmp/klippy_uds`

SkyShadex, 2022-11-02:
[image]

So I'm positive I should try to push another flash to the MCU with the current working setup.

But I'm looking at the `/dev/pts/` folder and just inspecting since atm I can't use termux and install tsu (repository down?). Of the times Ive plugged to device in it's either given me 3 or 2. I saw the tty under 3 so I assumed that's the one. And this it started spitting out this gibberish in the console. Forces me out after a while.

SkyShadex, 2022-11-03:
[klippy(4).log](https://github.com/d4rk50ul1/klipper-on-android/files/9925178/klippy.4.log)

Well I'm calling it for the night. Did succeed in getting moonraker and klipper talking. Re-Flashed, still no connection to the printer. Tried the following:

- `/dev/pts/0` 1-3 as well. 4 pops up occasionally
- `/dev/bus/usb/001/002` as read from octoprint
- all the tty variants.
[image]
 I don't have a `/dev/ttyACM0` or `/dev/ttyUSB0`.

The only one's that gave me any feedback were `/dev/pts/2` and 3. But I believe that's just it talking to itself given the random commands coming up that I didn't type and the repeating calls in the klippy.log.

I believe I fall into that category of having an android device that doesn't have an active ch341 driver? Moto X pure (XT1575). Now... the printer could just be flashed incorrectly, but I have no way of testing for that. It turns on and the fans come on is all I can say on that.

SkyShadex, 2022-11-03:
OMG
[image]

IT LIVES. I was just messing around with the cable because I figured this OTG thing might be an issue since pts/3 and pts/4 are present sometimes.

closed 2022-11-03 by SkyShadex

SkyShadex, 2022-11-04:
This might be conjecture. In power cycling the device and having the connection break, I'll notice the `/dev/pts/3` that I'm linking to disappears. I also notice that plugging into the printer sometimes has octoprint read `/dev/bus/usb/001/00*` Usually 2 or 3, sometimes 4. 

I've seen some success with running `sudo chmod 777 /dev/bus/usb/001/00*` whatever octoprint says.

Having troubles pushing enough power to sustain the device. Might have to get a beefier charger or make my own otg splitter.

Saurabh251-cmd, 2022-12-06:
> OMG [image]
> 
> IT LIVES. I was just messing around with the cable because I figured this OTG thing might be an issue since pts/3 and pts/4 are present sometimes.

can u share me config ,firmware.bin files? i got Ender 3 v2 too

SkyShadex, 2022-12-07:
https://github.com/SkyShadex/PrinterConfigs.git

As far as configs go, I started with the stock ender 3 profile and sort of worked my way up from there. If anything, test your stepper e-steps to make sure 100mm is a 100mm in each direction.

There's alot of borrowed macros and other additions that may or may not be messily commented

Saurabh251-cmd, 2022-12-07:
I have one request @SkyShadex Can you share me whole your setup folder inside xftp? Because Im having really hard time to setup

SkyShadex, 2022-12-07:
The rest of it is all just standard install stuff. Those are the only files I edit.

Where are you getting hung up?

The only places I had issues was correctly setting up Linux deploy (the slashes in the directory matter lol). 

Making sure in moonraker.conf, the host ip address at the top matches the ip address in Linux deploy. That way you can actually get access via app.fluidd.xyz

Once you can access moonraker, Figuring out the serial port. Octo4a + termux to find what it should be.

Saurabh251-cmd, 2022-12-07:
hmmmm i didnt know about changing host ip in moonraker tho ,Oke i will change and Which one should i go Fluid or mainsail?

Saurabh251-cmd, 2022-12-07:
ok buddy, what is use of outlook here?

SkyShadex, 2022-12-07:
Sorry, that was because I responded via email from phone instead of from github

Saurabh251-cmd, 2022-12-07:
oops its goood man,Ok so i will keep in contact with u until i run klipper without issue.
btw Thank you so much for helping me.

Saurabh251-cmd, 2022-12-07:
btw do u think its gonna work for otg and charging at same time?
[image]

SkyShadex, 2022-12-07:
Looks similar to the one I got. Check to make sure your device can actually charge while otg. It would be a kernel level thing.

Saurabh251-cmd, 2022-12-07:
ohhh i heard we can make mod for it so yeh ,if its about kernel i'm gonna give up for now then lol 
btw octoprint driver does make sense in case if smartphone don't support CH341 in kernel

Saurabh251-cmd, 2022-12-07:
im trying to reinstall screen and now this issue- 
[image]
:(

SkyShadex, 2022-12-09:
Would likely try a clean install from scratch. I never ran into issues with klipperscreen but I also rarely actually run it since fluidd is up on my desktop and phone if I'm home printing anyway.

Saurabh251-cmd, 2022-12-10:
aight i will try

Saurabh251-cmd, 2022-12-10:
Where to add this in my moonraker?-
[machine]
validate_service: False
validate_config: False
provider: none

Saurabh251-cmd, 2022-12-11:
seems evrything fine with klipper and smartphone but issue with config ig
hmmm
[image]

Saurabh251-cmd, 2022-12-11:
Finally got it working,Thanku but no screen on android tho

lucky13820, 2024-10-07:
I've got everything up and running except it won't connect to my printer. I checked use /dev/pts/2 in the mcu serial, but it doesn't seem to work.