d4rk50ul1/klipper-on-android #13: Unable to git clone without sudo

Kelsch, 2022-11-13:
I am pretty green behind the ears when it comes to anything with linux, so sorry if this is an easy problem. 

I have a pixel 4a that is rooted with magisk and running lineage OS.

My problem is that I can't git clone anything without sudo. If I try git clone https://github.com/th33xitus/kiauh.git I get an error saying Cloning into 'kiauh'...
fatal: unable to access 'https://github.com/th33xitus/kiauh.git/': Could not resolve host: github.com

Of course I can run this in sudo and it works fine, but then I run into the next problem of ./kiauh.sh trying to install klipper without sudo and that has the same issue.

Any ideas? Any help is appreciated

HouseDrVenus, 2022-12-13:
I have same problem with Xiaomi Mi 9T smartphone. I too, haven't figured out how to bypass this problem.

Vimknight, 2023-03-17:
Experiencing this same issue on a galaxy S8+ with LineageOS

Luffier, 2023-07-23:
Changing the user id for the user 'android' to 4000 fixed this for me. You need to run the following as root:

```shell
usermod -u 4000 android

find / -user 4000 -exec chown -h android {} \;
```

kcajjones, 2024-08-18:
> Changing the user id for the user 'android' to 4000 fixed this for me. You need to run the following as root:
> 
> ```shell
> usermod -u 4000 android
> 
> find / -user 4000 -exec chown -h android {} \;
> ```

This doesn't work for me as it states that "user is currently used by process: xxxxx".

PixelGames987, 2025-02-16:
> > Changing the user id for the user 'android' to 4000 fixed this for me. You need to run the following as root:
> > usermod -u 4000 android
> > 
> > find / -user 4000 -exec chown -h android {} \;
> 
> This doesn't work for me as it states that "user is currently used by process: xxxxx".

Had the same error, resolved it by doing passwd root as the "android" user, connecting with ssh root@<ip address> and running the command there.