d4rk50ul1/klipper-on-android #15: Write access disabled

mkxwell, 2022-11-20:
I used this script on a rooted phone inside a debian container with linux deploy (arm64, installation type: file) but I can not modify e.g. the printer.cfg file within the browser (by using fluidd or mainsail). Directories in the file system are owned by the android users.

[image]

moonraker.log says write access is disabled:
[image]

Any ideas?

Saurabh251-cmd, 2022-12-13:
U can change permissions ,just use xshell and login then type sudo su then chmod ugo+rwx file path press enter then exist from sudo su by (ctrl +d). make sure to use xftp for changing config etc and copy path of that file by clicking right click

Kenajcrap, 2023-03-22:
@mkxwell have you eventually been able to fix this? I'm experiencing the same problem

Kenajcrap, 2023-03-22:
fixed it, just follow the steps in the [troubleshooting section](https://github.com/d4rk50ul1/klipper-on-android#troubleshooting-ongoing-section-based-on-comments)

Trauer, 2023-09-04:
Same problem. Im not able to edit config files via mainsail or fluidd texteditor. The folders are all chmod 777 and owner android. its also not possible to upload gcode files or create folders. the steps in troubleshooting didnt help. Im on a Poco X3 NFC running rooted android 12

askolds11, 2023-09-20:
> fixed it, just follow the steps in the [troubleshooting section](https://github.com/d4rk50ul1/klipper-on-android#troubleshooting-ongoing-section-based-on-comments)

By adding
```
[machine]
validate_service: False
validate_config: False
provider: none
```
I was able to upload files, but not edit the .cfg files.
I then sudo nano'd the printer.cfg file and magically everything started working... Haven't set everything up yet, but at least the file editing works.

*of course, mainsail.cfg is still read only as per design