d4rk50ul1/klipper-on-android #16: Klipperscreen and Hostname error

kahupihu, 2022-12-04:
Hi!
I followed this guide on my rooted Lenovo K3 Note k50a40 with Lineage OS 15.1 or something and on stock android 5 but in both cases klipperscreen exited with pip failure saying that there is some problem with a package and not pip itself. In both cases, when I entered my device's IP into a browser, fluidd showed up asking me to enter in the hostname which is default localhost and doesn't work.

What should I do to fix this?
Btw, the hostname change in kiauh exits with:-

[✓ OK] Dependencies installed!

###### Creating backup of hosts file ...
[✓ OK] Backup done!
[✓ OK] File:'/etc/hosts.221204-1410.bak'

###### Setting hostname to 'ender3' ...

###### Please wait ...
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to create bus connection: Host is down

Please HELP!!

d4rk50ul1, 2022-12-09:
need more details to review..

stayingarmed, 2022-12-10:
I'm not the original poster, but I'm having the same issue and here is the code that shows during the error:

 ```
Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting setuptools
        Using cached setuptools-65.6.3-py3-none-any.whl (1.2 MB)
      Collecting wheel
        Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
      Collecting pyproject-metadata>=0.5.0
        Downloading pyproject_metadata-0.6.1-py3-none-any.whl (7.4 kB)
      Collecting typing-extensions>=3.7.4
        Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
      Collecting tomli>=1.0.0
        Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
      Collecting packaging>=19.0
        Using cached packaging-22.0-py3-none-any.whl (42 kB)
      Building wheels for collected packages: ninja, patchelf
        Building wheel for ninja (pyproject.toml): started
        Building wheel for ninja (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for ninja (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [9 lines of output]
            Traceback (most recent call last):
              File "/tmp/pip-build-env-c837fi5c/overlay/lib/python3.7/site-packages/skbuild/setuptools_wrap.py", line 612, in setup
                cmkr = cmaker.CMaker(cmake_executable)
              File "/tmp/pip-build-env-c837fi5c/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 148, in __init__
                self.cmake_version = get_cmake_version(self.cmake_executable)
              File "/tmp/pip-build-env-c837fi5c/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 105, in get_cmake_version
                ) from err
      
            Problem with the CMake installation, aborting build. CMake executable is cmake
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
        Building wheel for patchelf (pyproject.toml): started
        Building wheel for patchelf (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error
      
        × Building wheel for patchelf (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [9 lines of output]
            Traceback (most recent call last):
              File "/tmp/pip-build-env-8axh2jq_/overlay/lib/python3.7/site-packages/skbuild/setuptools_wrap.py", line 612, in setup
                cmkr = cmaker.CMaker(cmake_executable)
              File "/tmp/pip-build-env-8axh2jq_/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 148, in __init__
                self.cmake_version = get_cmake_version(self.cmake_executable)
              File "/tmp/pip-build-env-8axh2jq_/overlay/lib/python3.7/site-packages/skbuild/cmaker.py", line 105, in get_cmake_version
                ) from err
      
            Problem with the CMake installation, aborting build. CMake executable is cmake
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for patchelf
      Failed to build ninja patchelf
      ERROR: Could not build wheels for ninja, patchelf, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

#=======================================================#
 KlipperScreen installation failed! 
#=======================================================#

$ 

```

allherg, 2022-12-11:
You need to install a newer cmake (at least 3.15) since debian repository provides only 3.13. https://blog.eldernode.com/install-cmake-on-ubuntu-and-debian/

kahupihu, 2022-12-12:
Thank you! I will try this and report back to you!
The second problem I was having was when setting up fluidd. When opening the IP of the phone after restart, fluidd asks for a website URL and I have no idea what to put in there as anything I put gives a network error or CORRS Policy error.

kahupihu, 2022-12-12:
error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [289 lines of output]
      Collecting meson-python>=0.8.1
        Using cached meson_python-0.11.0-py3-none-any.whl (63 kB)
      Collecting meson>=0.60.0
        Using cached meson-0.64.1-py3-none-any.whl (895 kB)
      Collecting ninja
        Using cached ninja-1.11.1.tar.gz (27 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting patchelf
        Using cached patchelf-0.17.0.0.tar.gz (166 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Collecting setuptools
        Using cached setuptools-65.6.3-py3-none-any.whl (1.2 MB)
      Collecting wheel
        Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
      Collecting tomli>=1.0.0
        Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
      Collecting pyproject-metadata>=0.5.0
        Using cached pyproject_metadata-0.6.1-py3-none-any.whl (7.4 kB)
      Collecting typing-extensions>=3.7.4
        Using cached typing_extensions-4.4.0-py3-none-any.whl (26 kB)
      Collecting packaging>=19.0
        Using cached packaging-22.0-py3-none-any.whl (42 kB)
      Building wheels for collected packages: ninja, patchelf
        Building wheel for ninja (pyproject.toml): started
        Building wheel for ninja (pyproject.toml): still running...
        Building wheel for ninja (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error

        × Building wheel for ninja (pyproject.toml) did not run successfully.
        │ exit code: 1
        ╰─> [236 lines of output]

            --------------------------------------------------------------------------------
            -- Trying "Ninja" generator
            --------------------------------
            ---------------------------
            ----------------------
            -----------------
            ------------
            -------
            --
            Not searching for unused variables given on the command line.
            CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
            -- Configuring incomplete, errors occurred!
            See also "/tmp/pip-install-feqse54y/ninja_2177dbc41e134874b09a4dc0079e62fc/_cmake_test_compile/build/CMakeFiles/CMakeOutput.log".
            --
            -------
            ------------
            -----------------
            ----------------------
            ---------------------------
            --------------------------------
            -- Trying "Ninja" generator - failure
            --------------------------------------------------------------------------------

[... 194 lines cut ...]

        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
        Building wheel for patchelf (pyproject.toml): started
        Building wheel for patchelf (pyproject.toml): still running...
        Building wheel for patchelf (pyproject.toml): finished with status 'done'
        Created wheel for patchelf: filename=patchelf-0.17.0.0-py2.py3-none-linux_aarch64.whl size=89520 sha256=5e63832ac7eca6c309c31b6ff160882b743312738233f0ec1ff6981cdf5963c7
        Stored in directory: /home/android/.cache/pip/wheels/d0/01/b7/c41a88c23b3dd0deaa1c686c725cdccb5f14e557c98587ba66
      Successfully built patchelf
      Failed to build ninja
      ERROR: Could not build wheels for ninja, which is required to install pyproject.toml-based projects
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: pip install exited with status code 0
Trying again with new tools...

And after this, it fails again... :(

Saurabh251-cmd, 2022-12-13:
> error: subprocess-exited-with-error
>
> × pip subprocess to install build dependencies did not run successfully. │ exit code: 1 ╰─> [289 lines of output] Collecting meson-python>=0.8.1 Using cached [...]

try to remove klipperscreen from menu like do 2 times then againt try to install and avoid upper error,Just matter at last its supposed to written as KlipperScreen installation success

kahupihu, 2022-12-13:
Debian 11 stable worked for me "easier method"

Saurabh251-cmd, 2022-12-13:
> Debian 11 stable worked for me "easier method"

glad to hear

kahupihu, 2022-12-13:
I used kiauh only...

Saurabh251-cmd, 2022-12-13:
hmmm ig then its about setup 
environment issue

Saurabh251-cmd, 2022-12-13:
iseeee, Does your pts driver number changes after evry restart of phone?

kahupihu, 2022-12-14:
I use tcpserialusb app (tcpuart)... So yes

kahupihu, 2022-12-14:
https://github.com/gaifeng8864/klipper-on-android/issues/2#issuecomment-1333007469

Saurabh251-cmd, 2022-12-14:
> [gaifeng8864/klipper-on-android#2 (comment)](https://github.com/gaifeng8864/klipper-on-android/issues/2#issuecomment-1333007469)

Thank you so much bud,
Does we have to make changes in config and put command in terminal always after Evry phone restart?

Saurabh251-cmd, 2022-12-14:
Ah that's kinda work then like restart phone, make changes in config them put command then operation starts

kahupihu, 2022-12-14:
Y can create a startup script in Debian

Saurabh251-cmd, 2022-12-14:
> Y can create a startup script in Debian

U got?
BTW I need your help to setup tcpuart and camera setup for klipper
Ender 3 v2

kahupihu, 2022-12-14:
Im pretty sure contrab should work
As for the app, the apk is available online and the baid rate is 250000 and the type should be server with port default (8080)
Use an IP Camera app from play store and use the url http://ip:port/video in fluidd as a camera with mjepg adaptive. The port of the camera I use is 8083.

Saurabh251-cmd, 2022-12-14:
> Im pretty sure contrab should work As for the app, the apk is available online and the baid rate is 250000 and the type should be server with port default (8080) Use an IP Camera app from play store [...]

Im sorry contrab mean? like I wanna auto start that command in debian.
how u usually starts with these?

kahupihu, 2022-12-14:
*crontab
Search it up on Google...
It'll help you better than me :P

Saurabh251-cmd, 2022-12-14:
hehe thanks so u using that ?

kahupihu, 2022-12-14:
Idk if it works... 
Rn I'm trying to solve connection issues during prints

Saurabh251-cmd, 2022-12-14:
oh okay,Thanks