d4rk50ul1/klipper-on-android #17: Linuxdeploy "configuring core/locale..." reboot

gaifeng8864, 2022-12-07:
Hey,@d4rk50ul1
Have you encountered the following problem when installing debian through linuxdeploy?
When running to "configuring core/locale... ", my phone will restart automatically, and the installation fails.
I tried to change the installation path and installation source, but the result is the same, as long as this step is reached, it will automatically restart. Have you encountered or know of a solution to this problem?
[image]

d4rk50ul1, 2022-12-09:
Honnestly not, my linux deploy installation has worked properly in all my installations so far.

gaifeng8864, 2022-12-10:
ok thanks. Looks like I'm going to spend some more time trying to figure out why.

rogenth, 2023-04-08:
could you figure this out? I think it is a problem with the ssh installation

gaifeng8864, 2023-04-09:
> could you figure this out? I think it is a problem with the ssh installation

According to my test, it is caused by incompatibility between linuxdeploy and Android kernel version. At present, the solution to this problem is to replace other versions of the Android system, especially pay attention to the kernel version. Specifically, you need to test it yourself, and there is no recommended version number.

rogenth, 2023-04-10:
I managed to run everything through Termux now... that worked with the scripts of AnLinux and then sudo ./start-debian.sh, from there, all the options from linux deploy need to be configured manually.

gaifeng8864, 2023-04-11:
> I managed to run everything through Termux now... that worked with the scripts of AnLinux and then sudo ./start-debian.sh, from there, all the options from linux deploy need to be configured manually.

Is it working right now?

rogenth, 2023-04-11:
Yes, one old one plus x up and running with Fluidd, Moonraker and Klipper. Haven't tested Klipper Screen yet. Have to still mod the kernel for OTG charging, but I think I will just remove the battery since it's a hazard...

gaifeng8864, 2023-04-14:
> Hey, I wrote a little tutorial about my steps on Termux, please check it out: https://github.com/rogenth/klipper-on-android!

Great work, very nice. I have two questions for you: Does Octo4a need to be turned on all the time (such as booting up) to ensure that /data/data/com.octo4a/files/serialpipe is available? In addition, is it really feasible for OTG+Charge to use resistors to deceive mobile phones to achieve OTG+Charge?

rogenth, 2023-04-14:
Yes, the app of Octo4a needs to be on, but the Octoprint Server can be switched off. You may need the give permissions again every time you boot your device though, I haven't tested this more. 

Regarding OTG+Charge with resistors, I haven't done any physical modifications like this yet. The kernel can be tricked though into allowing OTG+Charge, that's what I did by modifying the Kernel with https://github.com/sollapse/opo_dwc3_otg. That allowed me to charge and use the OTG on a device that didn't allow this originally.

gaifeng8864, 2023-04-14:
> Yes, the app of Octo4a needs to be on, but the Octoprint Server can be switched off. You may need the give permissions again every time you boot your device though, I haven't tested this more. [...]

Thank you very much for your answer. Is there a general version of this kernel patch or is there a general cracking method?

gaifeng8864, 2023-04-14:
> Not that I am aware about. I remember just following a more general tutorial for compiling an android kernel. That kernel mod was for another phone with the same SoC (IOneplus One) and it worked well with the One Plus X. The patch itself for the OPO was forked from a project for another phone (Nexus 4/7) and they had to modify it (DWC3 USB driver handles the MSM8974 USB controller).

okay, thank you. It seems that this kernel patch and cracking method is not universal.

d4rk50ul1, 2023-04-14:
Hi guys, my updates regarding this implementation are that i stepped aside from it because my lenovo tab install got corrupted after a while and my prints started to fail. seems like running 24/7 on an entry class kind of tablet or smartphone doesn't seem to be a reliable long term solution. I ended up buying a cheap intel compute stick on ebay and mounted debian+klipper on it using kiauh and just used the same tabled enabling klipperscreen on it and, well, works like a charm i must say.

gaifeng8864, 2023-04-15:
> Hi guys, my updates regarding this implementation are that i stepped aside from it because my lenovo tab install got corrupted after a while and my prints started to fail. seems like running 24/7 on [...]

Haha, bad news, but also good news. Which Intel Compute Stick did you use? Is klipperscreen connected to the mobile phone using the usb interface?

rogenth, 2023-04-16:
> Hi guys, my updates regarding this implementation are that i stepped aside from it because my lenovo tab install got corrupted after a while and my prints started to fail. seems like running 24/7 on [...]

That seems to be a better option. The price of the stick is around 100 euros where I live, for that you can also get 2 orange pi. Maybe the corruption had to do with Linux Deploy? Would be interesting to see how a proot distro with Termux would work with this configuration.

d4rk50ul1, 2023-04-17:
> Haha, bad news, but also good news. Which Intel Compute Stick did you use? Is klipperscreen connected to the mobile phone using the usb interface?

Yeah, i connected the tablet and used klipperscreen only on it. As i said, everything works like a charm.

d4rk50ul1, 2023-04-17:
> That seems to be a better option. The price of the stick is around 100 euros where I live, for that you can also get 2 orange pi. Maybe the corruption had to do with Linux Deploy? Would be [...]

If you manage to test it out and validate, it might be handy. To your point regarding linux deploy, i would say yes. It constantly requests root permissions which for me indicates that it kind of heavily hits storage.... i whish we could just install debian or something locally over android without the struggle but well, it's just how products are built.