d4rk50ul1/klipper-on-android #22: My pts number always changes for MCU after every restart of phone

Saurabh251-cmd, 2022-12-13:
So basically my pts number always changes for MCU after every restart of phone.how to fix this?

d4rk50ul1, 2022-12-22:
It happens to me as well, everytime i restart i must update printer.cfg with the new address. Haven't gone over a fix for it @ZerGo0  any ideas?

ZerGo0, 2022-12-22:
Hey, I had the same issue and I fixed it by running a bash script when Linux Deploy is booted. I'm pretty sure that you could probably do all of this in the Linux Deploy Container itself, but this is how I'm doing it right now:

1. SSH into termux
2. Run the command to get the printer pts id
3. Set the id in the config file
4. Start/Restart Klipper

I'm not home right now and can't give you the script, but it's not too complicated. I have everything automated with tasker including powering on the printer via a Tasmota socket to get the printer pts id, adb wifi debugging, setting fake battery stats and so on. I'm too busy to do a right up of this tho.

Saurabh251-cmd, 2022-12-22:
Thanku so much, I will try but it will be good too if you share more information when you free :)

d4rk50ul1, 2023-01-27:
@Saurabh251-cmd  check the updated documentation. we found an easier way to keep octo4a and linux deploy in sync!