d4rk50ul1/klipper-on-android #24: klipper is failing to connect to CH340 with driver from `Octo4a`

bazarovdev, 2022-12-29:
Thanks for your work! 

It is my first try with klipper, I was able to connect from PC, run klipper and control printer from OctoPrint (connected to `/tmp/printer`)

Now I'm trying to run it from Nexus 10, installed LineageOS 13 (Android 6.0.1).
Octo4a 1.1.2 detects printer (but OctoPrint's installation is corrupted)
`/dev/bus/usb/002/002`
`/data/data/com.octo4a/files/serialpipe` points to `/dev/pts/2`

in `klippy.log` I get following log. Any ideas are welcome, thanks!

```
...
=======================
Declaration of '__main __octoprint' hides previous menuitem declaration
Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/android/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/android/klipper/klippy/serialhdl.py", line 259, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/android/klipper/klippy/serialhdl.py", line 317, in get_response
    cmd_queue)
  File "/home/android/klipper/klippy/serialhdl.py", line 251, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/android/klipper/klippy/serialhdl.py", line 61, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
...
```

bazarovdev, 2023-01-02:
Testing with scope and `minicom` I saw that the `/dev/pts/#`  actually "works" but:
```minicom -D /dev/pts/0 -b 250000``` (open `/dev/pts/0` at 250000 bps)
**TX:** my data sent at `112500` bps and each key press in `minicom` at `Octo4a` logs `115200`
**RX:** printer sends every few sec some msg at `250000`

Trying to run `klippy.py`:
```python ./klippy/klippy.py printer.cfg -l klippy.log```
**TX:** something is sent and looks like `250000` but `Octo4a` log constantly switches between `115200` and `250000`
**RX:** mcu replies at `250000` but feels at lower rate and `klippy.log`: `mcu 'mcu': Timeout on connect`

dsafak, 2023-01-07:
I am trying with ender 3 v2 and my phone also doesn't have drivers so using octo4a.
octo4a detects printer, I check with thermux it is pts/0 chmod etc. are done, printer config is set but in my case klipper also cannot connect.

bazarovdev, 2023-01-16:
Thanks @dsafak,
I tried to debug it and found something that seems as a problem in Octo4a, created [issue there](https://github.com/feelfreelinux/octo4a/issues/347).
Switching baudrate seems as good workaround that worked for me too.

closed 2023-01-16 by bazarovdev

TheAGunner, 2023-03-18:
I think I have the same communication problem with Octo4a and my SKR2, I tried changing the printer.cfg baudrate to 115200 as you said. Which communication interface did you use to make menuconfig for 115200 bauds? Will it work over the same usb cable?

alexbazz, 2023-03-18:
The flashing procedure doesn't work from Android, so I connected to PC, on PC in terminal run all the ’make menuconfig’ commands to configure, build and flash printer, and then went back to Android and it worked.

bazarovdev, 2023-03-18:
The flashing procedure doesn't work from Android, so I connected to PC, on PC in terminal run all the ’make menuconfig’ commands to configure, build and flash printer, and then went back to Android and it worked.

TheAGunner, 2023-03-19:
I appreciate the fast response! To flash the board I used a Raspberry Pi 4 with klipper working and selected communication interface "USB on PA11/PA12" on the make menuconfig. Octo4a is recognizing the board as "stm32f429", but I tried all of the available drivers and got no response from klipperscreen on the phone. Also, I gave permission '_sudo chmod 777 /home/android/octo4a/serialpipe_' from ssh into the phone. What should I check? Did you flash the board selecting Serial or Uart communication interface?

lucky13820, 2024-10-07:
> I am trying with ender 3 v2 and my phone also doesn't have drivers so using octo4a. octo4a detects printer, I check with thermux it is pts/0 chmod etc. are done, printer config is set but in my case klipper also cannot connect.
> 
> I checked klipper log and looks about same.
> 
> Okay I got it to work. I set firmware baud rate to 115200 and klipper to 115200 as well. [mcu] serial: /dev/pts/0 baud: 115200
> 
> I did a reflash so I am not sure if it worked because of this or re-flash. In any case this is how I solved it.

How do you set firmware baud rate to 115200? I don't have this option when I do `make menuconfig`

ivesiure, 2026-09-06:
Alternative approach, in case anyone still lands here from a search.

I hit this exact symptom — `Timeout on connect` during `identify_response`, CH340 at 250000 baud — and ended up going around the Octo4a driver entirely rather than trying to fix it.

The root cause on my device was that the Android kernel ships no `ch341` module at all, so there is no `/dev/ttyUSB*` for anything to open. Rather than recompile the kernel, I wrote a userspace CH341/CH340 driver in pure Python: `ioctl` straight on `/dev/bus/usb`, no kernel module, no libusb, no pyusb. It exposes the port as an ordinary pty, which Klipper opens as if it were a serial device.

https://github.com/ivesiure/ch341-userspace-pty

Two things turned out to be specifically necessary to get `identify` to complete at 250000 baud. Posting them here because they apply regardless of which route you take:

**The pty has to be in RAW mode on both ends.** If it isn't, the line discipline eats `0x11` and `0x13` as XON/XOFF — and Klipper's sequence byte is `0x10|n`, so `0x11` shows up constantly in normal traffic. The connection then fails in a way that looks like a baud rate or timing problem.

**`baud:` in `[mcu]` must be `115200`, not `250000`.** The rate configured on the pty side is fictional — there is no UART there. The real rate is set by the driver, on the chip. pyserial refuses non-standard rates on a pty and raises `NotImplementedError: non-standard baudrates are not supported on this platform`, which reads like a fatal error but is not one.

It has since run a 36-minute and an 86-minute print, both completed, with `Timer too close: 0` and `srtt` around 5–7 ms.

Licence is GPL-2.0-only, because the chip initialisation is translated from the kernel's `drivers/usb/serial/ch341.c`.

One caveat worth stating plainly: this has been tested against exactly one chip, `1a86:7523` version `0x31`, on one phone. The rest of the CH341 family shares the same init sequence in `ch341.c`, but I have not run it on anything else. Reports from other variants would be genuinely useful, including reports that it does not work.