d4rk50ul1/klipper-on-android #27: mcu: "Unable to connect" error

TheAGunner, 2023-02-19:
Hi! I was impressed with your work on this topic so I tried to power a setup the octo4a way (using a Lenovo Tab M10 plus 3rd gen and a BTT SKR2 board), I selected the CH341 driver on the app, changed the mcu serial in printer.cfg to "/home/android/octo4a/serialpipe" and set a 250k baudrate. Klipperscreen is launching, fluidd is accessible through browser but I still get the mcu: "Unable to connect" error. I already factory reseted 2 times and I'm still stuck at this point. Also, I get "Internal server error" on fluidd when trying to save changes on any .cfg file. Any suggestions on what to look for? Greetings.

d4rk50ul1, 2023-02-20:
Hi! glad you found us, we're trying to do our best here!
i think you're rushing to solve everywhere with a lot of places being modified at the same time! lol.
first: when you open OCTO4A and connect your printer, does it show a prompt asking for permissions to connect to your printer?
Second: when you say you add /home/android/octo4a/serialpipe, did you mount it on linux deploy as a resource?
third: my original tutorial was made with mainsail, so there might be some differences with the one made with fluidd which was the other link i added on top of my tutorial. you should check that one if you're followith fluidd path IMHO.
fourth: when you say internal server error issues on fluidd, where does this happen, in your pc screen or where?

TheAGunner, 2023-02-27:
Hi again, thank you for your fast response! Sorry I didn't notice until now.
1.	Yes, octo4A shows the prompt and I give the permission.
2.	Yes, I mounted it as you said in your tutorial.
3.	I wasn’t aware of it, I’ll go check it out!
4.	And yes, it appears in my pc browser (fluidd interface) when connecting to my tablet by IP address.
I’ll try the fluidd link and keep you updated, thank you for sharing your hard work!

TheAGunner, 2023-04-17:
Ok so after almost 3 months of testing, this morning I finally managed to get it to communicate correctly haha, klipperscreen showing and fluidd on browser.

[image]

I can confirm **Android 11** works for this method on my Lenovo M10 plus (3rd gen, Magisk rooted). 
For my **SKR 3 mini e3 v3**, the only driver that worked from octo4a was the 'CDC serial driver' (maybe other SKR boards difer). I had to specify 115200 baud on the printer.cfg [mcu] section (double check you don't have multiple declarations of this header xd), also the firmware on the board was flashed as if it were for a regular raspberry pi usb connection (no special changes here). 
The problem seemed to be on the chmod 777 permissions (thanks to [DalTron Printing](https://www.youtube.com/watch?v=B5gGMIXwDuo&t=1319s) response!, i suggest watching his video and custom klipper family bash script, he saved me from quiting this journey).
I had to start over again following [gaifeng8864](https://github.com/gaifeng8864/klipper-on-android/blob/main/README_en.md)'s way, but to make the final step to work (the bash script), using 'ls -al /dev/' was impossible for me to identify the board, so i manually replaced all 'ttyACM0' from the bash according to your solution proposed here [d4rk50ul1](https://github.com/gaifeng8864/klipper-on-android/issues/4#issuecomment-1405949370). 
Here was another problem, using FileZilla I discovered the real direction of the virtual mount done by the linuxdeploy app on the folder at ''/home/print3D/octo4a/serialpipe". There was only 1 folder and 1 remote access ('bootstrap' and 'serialpipe', again). I clicked on the remote access, and as you can see in the image, it was pointing at '/dev/pts/0':

[image]

So I replaced 'ttyACM0' with ''/home/print3D/octo4a/serialpipe/serialpipe", and the klipper family bash ran perfectly. Still, I had to 'sudo chmod 777 /dev/pts/0' manually from ssh after that because the fluidd web interface just prompted failure to connect on the logs. On the [mcu] header at printer.cfg, I used ''/home/print3D/octo4a/serialpipe/serialpipe".

[image]

Right now, my lectures are good, and the board info is displaying on the browser, but I still get the '**Internal Server Error**' when trying to save modifications of the printer config files directly from the fluidd web interface (i modified them externally transfering them using FileZilla), any suggestions on what I'm still missing?

[image]

Also, is it normal to get the 'fluidd-config', 'moonraker', and 'klipper' under software update with an error 'No valid, Recover' buttons? 

[image]

kdreamix, 2023-04-30:
I manage to get it to work too. Everything is working except I have to run `sudo chmod 777 /home/octo4a/serialpipe` everytime the phone is reboot. Seems like I lose permission after reboot, is there a way to setup a script to run chmod when
- phone is reboot AND
- wait for connection to octo4a serial port

TheAGunner, 2023-05-04:
There is a problem when I reboot the android, there are many prompts of apps that aren't running well right after the reboot, then the android reboots itself again and enters into a recovery screen: 'Can't load into Android system. Your data may be corrupt'.  Then a 'try again' or 'factory data reset' menu is below.

DavideBo05, 2025-07-26:
I have the same problem, but I didn't use octo4a, instead I used directly the serial /dev/ttyACM0 (verified by `ls /dev/` before and after unplugging the printer).
The fact is that sometimes it work, but if I restart the system, or if I disconnect and reconnect the printer, klipper no longer reaches it.
I'm not able to understand this... 
I tried also with the octo4a method, but `sudo chmod 777 /home/octo4a/serialpipe` returns me `chmod: cannot access '/home/octo4a/serialpipe': No such file or directory`, even if i mounted it in Linux Deploy ``