d4rk50ul1/klipper-on-android #28: Dosen't seem to work anymore

rolohaun, 2023-04-05:
I have a Lenovo M8 3rd gen with android 11 on it.

Moonraker will no connect to Klipper, cannot start Klipper service, I can not connect even though Octo4a shows connection to my printer.

TheAGunner, 2023-04-10:
Did you already try with the solution at this [issue](https://github.com/d4rk50ul1/klipper-on-android/issues/9)? Changing the mainsail.cfg line to "klippy_uds_address: /tmp/klippy_uds" worked for my Lenovo M10 plus (if you can't get to the file from browser, try using FileZilla), but I'm struggling getting klipperscreen to finally read the board, and although Octo4a is detecting the board, 'mcu error' connection keeps prompting. My Rook is waiting for this to work lol, great design btw!

rolohaun, 2023-04-10:
oh thanks! I will try that. Awesome, can't wait to see your serial request

TheAGunner, 2023-04-17:
Almost there!! I hope you find more useful information [here](https://github.com/d4rk50ul1/klipper-on-android/issues/27#issuecomment-1511754217).