d4rk50ul1/klipper-on-android #34: Documentation of solutions for problems that might occur

Luro02, 2023-09-11:
I spent the last few days trying to make this work on my Ender 3 S1 with a Xiaomi Mi Pad 4.

I searched a lot in the internet and this repo to try to figure out how to make this work. There are not a lot of resources, so I am sharing what some of my problems/solutions were. Maybe this will help someone else. Note that I was not yet able to establish a successful connection.

#### Compiling/Flashing the Klipper firmware

This guide does not mention how you can do this, the other linked one assumes that you already have a klipper firmware on your printer.

I was very confused why this is not mentioned, because the official klipper guide assumes you use a raspberry pi for this and the commercial variants like the sonic pad, compile the firmware for you automatically.

You can compile the firmware on your android device. Just follow this installation turorial until you have everything set-up except for the printer (execting the shell script).

You then connect to your android device through ssh. Your home directory should look something like this:

[image]

By the way, my shell looks like this, because I installed `zsh` through `sudo apt-get install zsh curl` and then [oh-my-zsh](https://ohmyz.sh/) through `sh -c "$(curl -fsSL https://raw.githubusercontent.com/ohmyzsh/ohmyzsh/master/tools/install.sh)"`

Anyway, you can then follow the [klipper guide](https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller) for compiling the firmware. To get the firmware bin, I recommend using filezilla which allows you to download files without installing an ftp server and has a graphical user interface. An alternative I can think of is to use scp, see [here](https://stackoverflow.com/a/9427585). The address you should connect to in filezilla is `sftp://your.android.ip`, use the credentials you set up and leave the port empty.

#### Printer not recognized in Octo4a with klipper firmware

Before installing the klipper firmware, I had the Ender3V2S1 CFW installed, which was recognized successfully by octo4a. After installing the klipper firmware, it was no longer recognized.

Some mention having this USB serial thing showing up in Octo4a

[image]

in which case one should be able to connect klipper through the app. This never showed up on my device, not even with the old firmware.

It turns out that your Android Kernel must have `CONFIG_USB_SERIAL` enabled for this to show. I am not 100% confident that this is the only Kernel option required, because some guides like [this one](https://all3dprint.blogspot.com/2021/12/compiling-kernel-for-android-arm64-with.html) mention other flags that are not present in my devices defconfig.

Some warnings/tips before you try to compile your android kernel yourself:
- Compiling your own android kernel is difficult, will take a lot of time and you might not be able to get it to work
- Do not even try to do this on windows/wsl. The Linux kernel has some files that windows file systems consider to be the same file. For example `ABC.txt` and `abc.txt` are the same file on windows, but not on linux. I recommend doing this in a virtual machine, trust me I tried it with WSL before. There is a workaround for this problem, but you will have many more problems later... It is easier to setup a VM that you can throw away after you are done.
- You do not have to compile your entire rom, just the kernel could be enough.
- After compiling the kernel successfully, you will end up with an `Image.gz` and `Image.gz-dtb` somewhere in your `out` folder. Those are important.
- For flashing I tried to unpack and repack the `boot.img`, but failed. Instead I had success with [`AnyKernel3`](https://github.com/osm0sis/AnyKernel3)

#### How do you check that the klipper firmware is working on your printer and the problem is your android device?

After compiling and flashing the klipper firmware, the printer was stuck on the "Creality" logo. I thought something was broken with my firmware, because others mentioned that the display should be dark.

You can check if the firmware flashed successfully by connecting your computer to the printer. On my Windows Laptop it showed up in the device manager as a (I think) CH340/CH341 COM device? If nothing is detected on your PC, you might have something broken?

On a Linux Computer, I would suggest just trying to install Klipper on it and connecting through it. Should be more reliable than trying to connect through an android device.

#### Somehow the printer is still not recognized?

Before I started this whole adventure I tested if I can connect to the printer and charge the device at the same time (not always possible with every device/adapter). This did work.

After flashing the klipper firmware/changing the kernel/cfw for the android device, it was no longer the case. So if you can not connect to your printer, it might be because you have the charger connected to the adapter at the same time.

#### The printer is recognized, but I can not connect to it?

This is the part where I am stuck myself. The device is only visible in Octo4a, so I followed the steps mentioned in this repository to pass through the connection to klipper. So far so good, the problem is that when I try to establish a connection to the printer in the web interface, I get a timeout/device is offline.

By looking through the Octo4a logs (top right of the screen), I noticed that it failed to read `/dev`. To fix this you have to set selinux to permissive. One way to do this, would be to change the kernel (but as mentioned above, I would not recommend doing this). Instead you can install the [`selinux_permissive`](https://github.com/evdenis/selinux_permissive) magisk extension, which should solve your problem.

To verify that it works successfully, I recommend connecting through the octo4a remote web terminal (you have to enable this in the app first) (enter in your browser: `your.android.ip:5002` with the default credentials being `root` and `octoprint` as password). In it you execute the `ls /dev` command. If it does return permission denied, selinux is not set to permissive.

It should look something like this (note that the list might have different names in it):

[image]

Luro02, 2023-09-11:
For now I reverted back to the Marlin firmware, but I will most likely try this again, once I have more time for this/do not require my printer for an extended time

ShuNegus, 2023-12-05:
@Pescu123 
I have a problem like you. Octo4a sees the printer and Klipper does not.  [The solution](https://github.com/gaifeng8864/klipper-on-android/issues/2#issuecomment-1333007469) from @gaifeng8864 helped me. Port forwarding with "tcpuart" apk works.

gaifeng8864, 2023-12-07:
> @Pescu123 I have a problem like you. Octo4a sees the printer and Klipper does not. [The solution](https://github.com/gaifeng8864/klipper-on-android/issues/2#issuecomment-1333007469) from @gaifeng8864 helped me. Port forwarding with "tcpuart" apk works.

It's great to be able to help you! I just don’t know what the principle of this software is. If possible, it would be best to integrate it into the system.