d4rk50ul1/klipper-on-android #35: Error when installing Moonraker

sisyphusSmiling, 2023-10-30:
Following the instructions as presented, I'm met with an error I can't seem to get past when installing Moonraker. While in the step creating the virtual environment, I get the following

```bash
Thanks buddy! Here's the error:

Building wheels for collected packages: pillow
  Building wheel for pillow (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [189 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-37
      creating build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageGrab.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImagePalette.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PixarImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/WalImageFile.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/JpegPresets.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageMorph.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageColor.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/MicImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PaletteFile.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PpmImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PdfImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/FitsStubImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/FtexImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/GimpGradientFile.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/_version.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageMode.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PcdImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/WmfImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/_binary.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/FontFile.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/MspImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/FliImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/FpxImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/features.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageTransform.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/MpegImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/SgiImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageShow.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/DcxImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PngImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageDraw2.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PSDraw.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/GdImageFile.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ExifTags.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/IcoImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/PalmImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/__init__.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/CurImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/JpegImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/EpsImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/BlpImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImageWin.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ContainerIO.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/ImagePath.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/_tkinter_finder.py -> build/lib.linux-aarch64-cpython-37/PIL
      copying src/PIL/QoiImagePlugin.py -> build/lib.linux-aarch64-cpython-37/PIL
[... 111 lines cut ...]
          raise RequiredDependencyException(f)
      __main__.RequiredDependencyException: zlib

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-td8leqae/pillow_1bc6a34b935246d0816f1495c788760c/setup.py", line 1010, in <module>
          raise RequiredDependencyException(msg)
      __main__.RequiredDependencyException:

      The headers or library files could not be found for zlib,
      a required dependency when compiling Pillow from source.

      Please see the install instructions at:
         https://pillow.readthedocs.io/en/latest/installation.html

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pillow
```

I confirmed that zlib is installed, and installed pillow directly from the home directory. Not sure where to go from here.

MrAngryStump, 2023-11-07:
i have absolutely same issue

Pescu123, 2023-12-04:
You could try to follow [this](https://github.com/python-pillow/Pillow/issues/4242) thread to troubleshoot your problem. You can see in the log that you send that the problem is this: 

> The headers or library files could not be found for zlib,
      a required dependency when compiling Pillow from source.

Chazuba, 2023-12-16:
I had the same problem, but problem was gone when i switched config of linux deploy to debian oldstable (was using arm64)

wischm0b, 2024-01-28:
> I had the same problem, but problem was gone when i switched config of linux deploy to debian oldstable (was using arm64)

That worked for me.